timo1235/cnc-werkstatt #1: Verständnisfrage RJ45

BennyBread, 2022-11-26:
Hallo Timo. Ich beschäftige mich gerade mit dem Board und der Verkabelung...

Die RJ-45 Adapterplatine dient ja zum Verbinden mit dem Treiber. Wo ich mir unschlüssig bin ist welches Signal jeweils die + (Plus) und - (Minus) Adern haben.. 

Wenn man als Beispiel ENA :
Pin1:  ENA+  = ENA  oder 5V?
Pin2:  ENA-   = ENA oder 5V?

Laut deiner Beschreibung auf der Doku Seite ist  Zitat: "STEP   => Signal für die Schritte. Manchmal auch **PUL(+)** genannt."
Aber laut dem Datasheet vom z.B. DM556 ist PUL+ mit VCC (+5V) verbunden und  PUL- mit dem PUL Signal vom Controller.
Siehe hier: [DataSheet](https://www.upload.sorotec.de/doku/manuals/DM556_data_sheet_de_131022.pdf)

Vielleicht denk ich nur zu kompliziert, aber wenn ich das Datenblatt richtig verstehe dann müsste "dein" STEP auf PUL-  (ohne Vorwiderstand bei 5V)

Danke schonmal für die Hilfe.
Gruß, Benny
PS: Super Sache dein Projekt, vollen Respekt; hast Dir echt ein paar gute Gedanken gemacht! Bestellung ist schon getätigt ;-)

BennyBread, 2022-11-26:
Öm.. OK. Hab jetzt mal das User Manual direkt von Leadshine herangezogen:

[Leadshine DM556E Manual](https://www.leadshine.com/upfiles/downloads/d19dd5aeb9ceb378d9a7882ab9551217_1651053026312.pdf)
4.1 Control Signal Connection. Was haben wir denn bei "Dir" Common Anode oder Common Kathode?

Sorry, bin grad verwirrt..

taltholtmann, 2022-11-29:
Hallo, 
es wird nach "common cathode" geschaltet. Also STEP an PUL+. Gib Bescheid, falls es nicht funktioniert.

BennyBread, 2022-11-30:
Danke für die Info.

closed 2022-11-30 by timo1235

BennyBread, 2022-12-23:
Sorry nochmal. An PUL- am Treiber kommt dann GND, oder? 🙈

timo1235, 2022-12-26:
Yes.