timo1235/cnc-werkstatt #2: Problem mit Autosquare / Eingang 8

Fendo99, 2022-12-30:
Folgendes Problem.
Eingang 8 wird nicht als geschaltet erkannt. Das führt dazu dass beim Autosquare der Stepper für Y2 auf Anschlag fährt und dann nicht aufhört zu stoppen.
Konfiguriert sind die Endstopps als NC. Auch beim bespielen des Arduino wurde auf HIGH gestellt. Die Eingänge 5 bis 7 funktionieren einwandfrei.
Beim Durchmessen ist mir aufgefallen dass bei den Pins am Arduino im geschalteten Zustand von Eingang 5 bis 7 etwa 4,6V anliegen. Bei Eingang 8 (Pin D6) allerdings nur 1,8V. Könnte es daran liegen dass er nicht schaltet? Wenn ich den Arduino abnehmen und ohne das Shield messe liegen an allen 4 Eingängen gleichmäßig etwa 4,6V an.
Hoffe mir kann da jemand helfen.

timo1235, 2022-12-31:
Du schreibst leider nicht, welche Steuerung du nutzt. Falls du einen Arduino Mega nutzt, kann das hier deine Lösung sein: https://blog.altholtmann.com/open-cnc-shield/dokumentation/#wieso-zeigt-eingang-8-in-estlcam-immer-ausgelst-an

Fendo99, 2022-12-31:
> Du schreibst leider nicht, welche Steuerung du nutzt. Falls du einen Arduino Mega nutzt, kann das hier deine Lösung sein: https://blog.altholtmann.com/open-cnc-shield/dokumentation/#wieso-zeigt-eingang-8-in-estlcam-immer-ausgelst-an

Ja. Ich nutze 2 Arduino Mega + Estlcam. Da ich die Eingänge Invertiert habe wegen der NC Endschalter ist mir das natürlich nicht aufgefallen dass dieser Eingang permanent geschaltet ist. Da es ja der Standard Zustand ist.
Ich vermute mal dass ich dann bei beiden Arduinos die LED entfernen muss oder?

timo1235, 2022-12-31:
Nein, nur der mit estlcam per USB verbunden ist.

closed 2023-02-12 by timo1235