timo1235/cnc-werkstatt #5: Endstop verhält sich merkwürdig

Findlerman, 2023-04-26:
Habe die Schaltung jetzt soweit,  dass ich sie testen kann. Leider macht mir der NPN Näherungsschalter etwas Kopfzerbrechen:
Wenn ich den Eingang auf "-" stelle (da der NPN ja GND auf die Signalleitung schaltet) reagiert die Steuerung nicht.

Wenn ich den Eingang jedoch auf  "+" stelle, dann bekommt die Steuerung das Schalten mit. 

Messe ich zwischen Signalleitung (nicht angeschlossen) und COM1 habe ich 24V (Boardspannung) bzw 0V - folglich schaltet der Senso auch GND.
Messe ich im eingebauten Zustand fällt die Spannung um etwa 5V ab (vermutlich Spannungsteiler für den Arduino? ).

Aber warum funktioniert es nur, wenn ich den Schiebeschalter in die falsche Position bringe?

[image]

[image]

timo1235, 2023-04-26:
Ja, das "Problem" kenne ich. Das hat was mit den verschiedenen potenzialen(5V, 24V) zu tun.
Der Schalter zum Wechsel von (-) auf (+) ist mit GND, bzw 5V verbunden. Müsste aber in deinem Fall mit 24V verbunden sein. 

Ich müsste da mal die Schaltung/Beschriftung ein wenig anpassen. Das hat aber keine Priorität, da es auch so ohne Probleme funktioniert. Einfach bei einem 24V NPN den Switch in (+) Stellung und dann in der Software den Eingang invertieren...

Bei 5V NPN funktioniert das richtig herum. Die meisten 6-30V Sensoren funktionieren auch mit 5V, kannste ja mal testen. Also an COM1 nur 5V und dann den Jumper in (-) Stellung. Das wird funktionieren, wie gewünscht. 

Ich würde den Sensor aber dennoch mit den vorgegebenen 24V an COM1 betreiben und den Jumper dann in (+) Stellung.

Findlerman, 2023-04-27:
Danke, für die schnelle Antwort. 
Wenn es problemlos so funktioniert, ist ja alles gut, dann kann ich jetzt unbesorgt weiter machen... 
Wäre nur hilfreich gewesen, wenn es in  einer kleinen Anmerkung in die Dokumentation Einzug gehalten hätte... 
Ich hab erstmal ne Stunde an mir selbst und an dem NPN Aufdruck auf dem Sensor gezweifelt und dann gedacht "schei** drauf, dann grill ich halt im schlimmsten Fall halt den Arduino, wenn ich den Schalter umlege, aber ich will es jetzt wissen!"... und danach wurde die Verwirrung dann noch größer ;)

HaWiWe, 2026-10-06:
Ich habe jetzt das neue Board von dir erhalten und die Anschlüsse identisch belegt wie vorher, jetzt funktioniert mein Sensor nicht mehr (der Knopf des Sensors leuchtet und geht aus, wenn er triggern sollte), es kommt aber nichts am Board an. Ist das das gleiche Problem?