timo1235/cnc-werkstatt #6: OCS2 Estlcam mit GRBL

Unuluminium, 2023-05-27:
Hallo. Ich habe mir das Estlcam-Modul geholt, dort stand, dass man mit angepasster Software GRBL nutzen kann.
Hast du dafür eine fertige Version oder ist die eine im Fork die richtige?
Ich habe es selbst versucht im VSC das Pinmapping anzupassen aber zb ENA gibt es ja nicht auf dem ESTLCAM-Modul.
Danke und Grüße

timo1235, 2023-05-29:
Heyy, 
ich habe damals diesen Fork erstellt und nutze den aktuell immer noch genauso an meinem Laser: https://gitlab.com/cnc-projects1/grbl

Hier ist noch eine andere Version, welche ich gefunden habe: https://github.com/Eisbaeeer/Laser_grbl_Mega2560

Unuluminium, 2023-05-29:
Servus Timo.Ich habe es direkt ausprobiert…Deine Software entspricht genau meiner.Das Problem ist, dass ich auf dem PWM-Kanal dauerhaft 0,330V habe.Der Laser ist immer an.Der Laser ist genau so angeschlossen wie am tillboard, nur jetzt mit einer externen 12V-Versorgung.Also nur der ttl geht vom PWM-Port ab.GrüßeVon meinem iPhone gesendet

Unuluminium, 2023-05-30:
Servus,
ich habe heute Nacht keine Ruhe bekommen.
Irgendwas ist Faul. Ich bekomme jetzt zwar einen PWM raus aber immer nur 5V.

Ist der Laser angeschlossen und ich gebe ein Signal, dann gehen von 5V der
externen Versorgung über das Mainboard.
Der Laser ist mit 5V auf 100%.

Ich habe ihn nochmal an das TIllboard geschlossen, dort funktioniert er
völlig normal.

Ist Schaltungstechnisch der PWM am OCS2 anders gelöst als am Tillboard oder
ist es womöglich ein Hardware-Fehler?

Grüße

Tony

timo1235, 2023-05-31:
Heyy, 

die Schaltung ist genau wie im Tillboard. Dein Text ist schwer nachvollziehbar. 
Dass der Laser bei 5V auf 100% ist, ist normal und richtig. 

Das Tillboard hat keinen PWM Terminal. Meinst du den Pin Header auf dem Board, wo man das PWM Signal abgreifen kann? Falls ja, dann funktioniert der PWM Ausgang am OPEN-CNC-Shield gleich.

Wie gesagt, ich hab es bei mir mit der oben genannten Software im Einsatz, daher würde ich darauf tippen, dass der Laser falsch angeschlossen ist. Du kannst ja mal Bilder posten, wie du den Laser angeschlossen hast.

Funktioniert die PWM Ausgabe richtig, wenn du mit Estlcam die Spindel startest? Das ist auch ein PWM Signal.

Unuluminium, 2023-05-31:
Servus Timo,
tut mir leid für die Verwirrung.
Der Laser ist 1:1 angeschlossen wie beim Tillboard.
Beim Tillboard geht er, beim OCS2 nicht.

Beim OCS2 greife ich den TTL eben vom PWM-Terminal ab.
Den GND habe ich auf den GND gesteckt:
Schalte ich jetzt den PWM, bricht die externe12V-Versorgung ein und 5V
laufen über den TTL auf den GND vom OCS2 zurück.

Im Anhang ein Bild. Ich habe alles mehrmals geprüft und auch die Funktion
nochmals am Tillboard getestet, es geht dort.

Grüße

Tony

Unuluminium, 2023-06-03:
Servus Timo,

verstehe mich bitte nicht falsch, ich benötige das Board und erzähle keinen
Mist.
Ich weiß auch wie man einen PWM misst oder wo was rauskommen sollte.

Du kannst mir gerne nochmal sagen zwischen welchen beiden Punkten du den
PWM misst und ich mache es 1:1 nach, ansonsten würde ich gerne nur das
Mainboard dir zuschicken, damit du dich selber davon überzeugen kannst.

Es wäre schön, wenn wir das schnell gelöst bekommen...

Danke und Grüße

Tony

timo1235, 2023-06-03:
Nabend,

Hast du Mal geschaut was du zwischen gnd und vcc misst? Der Anschluss ergibt sich aus dem pwm Ausgang. Der sollte sich also auch je nach Spindelgeschwindigkeit ändern.

Du kannst mir das Board jederzeit zurück schicken. Ich schaue dann kurz drüber und melde mich.

Viele Grüße
Timo

Unuluminium, 2023-06-03:
Hallo Timo,
es tut mir wirklich Leid aber ich springe hier im Dreieck, es geht plötzlich....

Ich habe auf dein anraten hin die Sterueung von Estlcam aufgespielt und den Spindle-PWM gemessen und das ging auf einmal.
Danach habe ich wieder die GRBL aufgespielt und es ging nicht mehr, danach habe ich einmal neu gestartet und jetzt geht zum ersten Mal der PWM über GRBL.

Ich habe nichts an der Verdrahtung geändert noch an der Software.

Ich stehe hier wie der letzte Depp da, ich kann weder ausschließen, noch einen Grund erkennen.
Betrachte diesen Issue als geschlossen und den Fehler 30cm vor dem Bildschirm sitzend.
Danke für deine Unterstützung und tut mir leid für mein gespamme.
Grüße
Tony

closed 2023-06-03 by Unuluminium