timo1235/cnc-werkstatt #8: Fertigungseinstellungen für PCB

Chrisdor1, 2025-02-01:
Hallo Timo

Ich möchte die Platinen für den Wireless Adapter und Wireless Panel bei JLCPCB bestellen, bin mir aber mit den Einstellungen für die Fertigung nicht sicher. Könntest du einen kurzen Blick auf die Angaben machen, ob das so passt?

Lg. Christoph

[image]

timo1235, 2025-02-02:
Moin.
Wenn du alles bei den Standard Einstellungen gelassen hast sollte das wohl passen.

Ich kann dir aber nahe legen, die neueren Varianten mit dem Esp32 zu nutzen, anstatt der Arduino mega lösung mit nrf24... 

Ansonsten könnte ich mal Zuhause schauen, ob ich noch Reste der Platinen finde. Dann brauchst du die nicht kaufen... Würde ich dir schenken. Kann ich aber erst morgen prüfen.

Chrisdor1, 2025-02-03:
Hallo,

ja hab alles bei den Standart Einstellungen gelassen.

Den Esp32 kann ich aber mit dem Tillboard und Tillboard Extension nicht nutzen oder? 
Dann müsste ich auf das Open-CNC-Shield 2 wechseln.

Ja falls du noch Platinen hast, Versand würde ich natürlich übernehmen.

Lg.

timo1235, 2025-02-03:
Leider habe ich nur noch die Wireless Adapter Platinen hier. 

Ja, das stimmt, du müsstest dann wechseln.

Chrisdor1, 2025-02-04:
Alles klar!

Umsteigen auf das CNC-Shield möchte ich zur Zeit nicht da das Tillbord + Extencion im Moment völlig ausreichend ist.

Danke für das Angebot bezüglich der Platinen, ich werde mir aber gleich beide Platinen bestellen.

timo1235, 2025-02-04:
Alles klar.

closed 2025-02-04 by timo1235